DavidAntliff/esp32-owb #6: Looks like some functions are being depricated

chmorgan, 2018-07-10:
CC build/esp_http_client/esp_http_client.o
/components/esp32-owb/owb_gpio.c: In function '_reset':
/components/esp32-owb/owb_gpio.c:100:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
     taskENTER_CRITICAL(&timeCriticalMutex);
             ^
/components/esp32-owb/owb_gpio.c:134:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
     taskEXIT_CRITICAL(&timeCriticalMutex);
             ^
/components/esp32-owb/owb_gpio.c: In function '_write_bit':
/components/esp32-owb/owb_gpio.c:156:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
     taskENTER_CRITICAL(&timeCriticalMutex);
             ^
/components/esp32-owb/owb_gpio.c:164:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
     taskEXIT_CRITICAL(&timeCriticalMutex);
             ^
/components/esp32-owb/owb_gpio.c: In function '_read_bit':
/components/esp32-owb/owb_gpio.c:177:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
     taskENTER_CRITICAL(&timeCriticalMutex);
             ^
/components/esp32-owb/owb_gpio.c:198:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
     taskEXIT_CRITICAL(&timeCriticalMutex);
             ^

DavidAntliff, 2018-07-15:
@chmorgan is this with the latest v3.0 or v3.1-beta1?

chmorgan, 2018-07-15:
Latest master from about five days ago. I tend to track master when I push
pull requests against esp-idf like I did earlier this week.

DavidAntliff, 2018-08-12:
I found that the new functions work with v3.0 so I've pushed 4629d1923869195e99256cdffb9adfc8ae51c982.

closed 2018-08-12 by DavidAntliff

chmorgan, 2018-08-13:
Cool thanks David!