DavidAntliff/esp32-owb #7: Scrach Pad size

povi2946, 2019-06-17:
This is more of a question as I noticed that it works both ways.

Why in this function raw rom byte size is 8 instead of 9? Isn't last 9th bit the crc bit?

ypedef union
{
    /// Provides access via field names
    struct fields
    {
        uint8_t family[1];         ///< family identifier (1 byte, LSB - read/write first)
        uint8_t serial_number[6];  ///< serial number (6 bytes)
        uint8_t crc[1];            ///< CRC check byte (1 byte, MSB - read/write last)
    } fields;                      ///< Provides access via field names

    uint8_t bytes[8];              ///< Provides raw byte access

} OneWireBus_ROMCode;

DavidAntliff, 2019-06-17:
If I assume you are talking about the DS18B20, take a look at the [datasheet](https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf) on page 8. There are two aspects to consider:

1. The 64-bit ROM code - this is eight 8-bit bytes (octets), where the LSB is the family code, the next 48 bits (6 bytes) is the serial number, and the last 8 bits (eighth byte) is the CRC. The struct that you pasted is this ROM code.

2. The scratchpad is nine bytes (see Figure 9) - and yes, the last byte is the CRC byte. But the code you pasted isn't the scratchpad struct, it's the ROM code struct.

The relevant struct is [here](https://github.com/DavidAntliff/esp32-ds18b20/blob/359cf918e3f641159283b5ae96cd939df28ef6e6/ds18b20.c#L62):

```
typedef struct
{
    uint8_t temperature[2];    // [0] is LSB, [1] is MSB
    uint8_t trigger_high;
    uint8_t trigger_low;
    uint8_t configuration;
    uint8_t reserved[3];
    uint8_t crc;
} Scratchpad;
```

This has a size of nine bytes.

Please let me know if this isn't clear.

povi2946, 2019-06-18:
I see. I am using AWS FreeRTOS which is compiled in C instead of C++, so I was using your library as reference and rewriting it for C and when adding CRC bit I checked OWB library portion. 
And inside owb_read_rom() function there is a call for 
` if (owb_crc8_bytes(0, rom_code->bytes, sizeof(OneWireBus_ROMCode)) != 0)`
where sizeof(OneWireBus_ROMCode is 8 and inside ds18b20_read_temp() function there is a line
`if (owb_crc8_bytes(0, buffer, 9) != 0)` which sets the data length to 9.
Since I was looking for CRC implementation example, I looked only at owb library and it confused me as in my code it works only with the length of 9.  I am working with MAX31850K and I know all addresses of the devices I made a very simple implementation of the library and overlooked few things from DS18B20 library.
Thank you very much for helping to clarify things for me and thank you for your work on this library.

DavidAntliff, 2019-06-18:
Fair comment - I should get rid of that `9` and use the actual struct size. Thanks for pointing that out.

BTW, this library is written for C & FreeRTOS. It is not a C++ library, although I do have some header guards in case someone does try to compile with C++. Therefore it should work with AWS FreeRTOS out-of-the-box. Also I believe someone has successfully used my DS18B20 library with the MAX31850: https://github.com/DavidAntliff/esp32-ds18b20/issues/4

povi2946, 2019-06-21:
Hi,
your demo worked perfectly for me, but when I was trying to include it into my AWS FreeRTOS demo I had issues compiling and when I was trying to debug I thought It was something because of C++. Anyways, it just shows how improficient I am. I remember it was something that it couldn't find definition for reset() function which was actually defined as _reset(). Anyways, I could have posted a proper debugging output but new AWS FreeRTOS got released with so many changes in their libraries, so I got occupied trying to migrate to new stuff. I hope in next couple days I will have a chance to give another try of adding your library. I really appreciate your help, thank you for your time.

DavidAntliff, 2020-03-15:
https://github.com/DavidAntliff/esp32-ds18b20/issues/5

closed 2020-03-15 by DavidAntliff