DavidAntliff/esp32-owb #12: owb_verify_rom blocks, if no one wire device is connected

marcel-behlau-elfin, 2020-01-27:
The while loop within the function owb_verify_rom blocks, if no one wire device is connected to the bus.

https://github.com/DavidAntliff/esp32-owb/blob/master/owb.c#L357

DavidAntliff, 2020-01-27:
Thanks for reporting. Are you using the RMT or the GPIO driver? Is the bus completely empty? What are you using for your pull-up resistor?

When the OWB Reset is issued, a `is_present` flag is only set to true if a device on the bus responds to the reset with a "presence" pulse. If there is no such pulse, then this flag remains false. This means that the `_search()` function finds no devices, and the `state->last_device_flag` flag is set false, terminating the while loop.

If you are seeing this loop spin infinitely, I'd be keen to hear more about your setup and the code that calls these owb functions, please.

marcel-behlau-elfin, 2020-01-30:
Hi David,
 
i'm using the rmt implemtation with GPIO_NUM_18. There is only the PullUp (4k7) and a protection diode (https://datasheets.maximintegrated.com/en/ds/DS9503.pdf) connected to the bus.
 
I assume the problem is related to the state of last_device_flag.
 
```
E (6448) owb: owb_verify_rom called
E (6448) owb: owb_verify_rom: Run: 0, last_device_flag: 0 is_found: 0
E (6458) owb: owb_verify_rom: Run: 1, last_device_flag: 0 is_found: 0
E (6468) owb: owb_verify_rom: Run: 2, last_device_flag: 0 is_found: 0
E (6468) owb: owb_verify_rom: Run: 3, last_device_flag: 0 is_found: 0
E (6478) owb: owb_verify_rom: Run: 4, last_device_flag: 0 is_found: 0
E (6488) owb: owb_verify_rom: Run: 5, last_device_flag: 0 is_found: 0
E (6498) owb: owb_verify_rom: Run: 6, last_device_flag: 0 is_found: 0
E (6498) owb: owb_verify_rom: Run: 7, last_device_flag: 0 is_found: 0
E (6508) owb: owb_verify_rom: Run: 8, last_device_flag: 0 is_found: 0
```

DavidAntliff, 2020-02-01:
I will look into this further.

In the meantime, you can do a search on the bus and if the number of connected devices is zero, avoid calling `owb_verify_rom()`. For example: https://github.com/DavidAntliff/esp32-ds18b20-example/blob/master/main/app_main.c#L58

DavidAntliff, 2020-02-01:
Also, can you describe where the diode is? Do you have a schematic diagram? I assume it works OK when a device **is** connected?

marcel-behlau-elfin, 2020-02-27:
Hi,
the connection is identical to Figure 4.

I also tried a different pin (GPIO4) and only a 4k7 pullup and was able to reproduce the problem.

As test application i modified your https://github.com/DavidAntliff/esp32-ds18b20-example/blob/master/main/app_main.c
```
diff app_main.c mod_app_main.c 
80,97c80,97
<     if (num_devices == 1)
<     {
<         // For a single device only:
<         OneWireBus_ROMCode rom_code;
<         owb_status status = owb_read_rom(owb, &rom_code);
<         if (status == OWB_STATUS_OK)
<         {
<             char rom_code_s[OWB_ROM_CODE_STRING_LENGTH];
<             owb_string_from_rom_code(rom_code, rom_code_s, sizeof(rom_code_s));
<             printf("Single device %s present\n", rom_code_s);
<         }
<         else
<         {
<             printf("An error occurred reading ROM code: %d", status);
<         }
<     }
<     else
<     {
---
>     //if (num_devices == 1)
>     //{
>     //    // For a single device only:
>     //    OneWireBus_ROMCode rom_code;
>     //    owb_status status = owb_read_rom(owb, &rom_code);
>     //    if (status == OWB_STATUS_OK)
>     //    {
>     //        char rom_code_s[OWB_ROM_CODE_STRING_LENGTH];
>     //        owb_string_from_rom_code(rom_code, rom_code_s, sizeof(rom_code_s));
>     //        printf("Single device %s present\n", rom_code_s);
>     //    }
>     //    else
>     //    {
>     //        printf("An error occurred reading ROM code: %d", status);
>     //    }
>     //}
>     //else
>     while(1){
```

Are you able to reproduce the problem in this way?

DavidAntliff, 2020-02-27:
Thanks for the extra info. I will try to reproduce this and get back to you, here.

DavidAntliff, 2020-03-18:
@marcel-behlau-elfin I think I've fixed this - can you update your `esp32-owb` component and see if it resolves the issue, please?

DavidAntliff, 2020-03-18:
Refer to https://github.com/DavidAntliff/esp32-ds18b20-example/pull/10 for use of this fix in the example.

marcel-behlau-elfin, 2020-03-30:
Thanks, works great!

closed 2020-03-30 by marcel-behlau-elfin