DavidAntliff/esp32-owb #16: Parasitic Power Mode

iotexpert, 2020-06-15:
David & @chmorgan 

Nice library thanks.

There is a comment in owb.h that says "Currently only externally powered devices are supported. Parasitic power is not supported." but the README.md says "Parasitic power mode."  However, my viewing of your code make it look like it is there... so my question is: Is which is correct?

If it doesn't work, I thought that I would fix it... but any insights into how you knew it wasn't working would be helpful as  a starting place.

Thanks,
Alan

DavidAntliff, 2020-06-15:
Hi, parasitic mode is supported, the comment is out of date. Thanks for letting me know.

DavidAntliff, 2020-06-15:
I’ll leave this open until I’ve fixed the comment.

DavidAntliff, 2020-06-15:
Fixed in 6f90bcdc4d7c29f3888f18e1baf74ad3116c823d - thanks again for the heads-up.

closed 2020-06-15 by DavidAntliff