DavidAntliff/esp32-owb #18: Documentation on owb_rmt_initialize

abraha2d, 2020-10-19:
The documentation for `owb_rmt_initialize` does not indicate that the `owb_rmt_driver_info` structure needs to be kept around, since the OneWireBus is actually a member of the info structure.

If the info structure goes out of scope, then things break down, and it's not apparent why. (I just spent a few hours tracking down an issue caused by this.)

DavidAntliff, 2020-10-19:
@abraha2d I'm sorry you spent time tracking this down when better docs would have helped you avoid this. Thank you for reporting it. I will update the docs in due course and hopefully prevent anyone else having the same experience.

abraha2d, 2020-10-19:
No worries, it was a good learning experience. Figured this will help others from encountering the same issue.

Thank you for the work with this one-wire driver, it's been very useful to me.

closed 2020-10-21 by abraha2d