DavidAntliff/esp32-owb #19: Bad device on bus causes ds18b20_check_for_parasite_power to hang

lukecyca, 2021-01-09:
I have a DS18B20 that is malfunctioning. When it is plugged in to the bus, the call to `ds18b20_check_for_parasite_power()` hangs forever. I believe it is hanging in its call to `_read_bits()` hence why I have opened the issue here.

Obviously my malfunctioning device is the root cause here. However I think even a misbehaving bus should not cause this call to hang indefinitely.

The debug output is:
```
D (3428) ds18b20: ds18b20_check_for_parasite_power
E (3528) owb_rmt: rx_items == 0
D (3528) owb_rmt: _is_present 0
D (3528) ds18b20: owb_reset OK
D (3528) ds18b20: owb_write_byte(ROM_SKIP) OK
D (3528) ds18b20: owb_write_byte(POWER_SUPPLY_READ) OK
[hangs]
```

DavidAntliff, 2021-01-11:
I've looked through the code and I can see a few places where a hang _might_ be occurring, but without a malfunctioning device it's going to be hard for me to track it down decisively. Since you have such a device, do you think you might be able to move around a few strategic logging statements to see if you can work out which loop is not terminating, or which IDF function call is not returning, please?

lukecyca, 2021-01-11:
It is hanging on this line: [owb_rmt.c:284](https://github.com/DavidAntliff/esp32-owb/blob/6b544d89fbffed26a7b4d85c2acc4d3bc329af5f/owb_rmt.c#L284)
```
        rmt_item32_t *rx_items = (rmt_item32_t *)xRingbufferReceive(info->rb, &rx_size, portMAX_DELAY);
```

I'm assuming the fix is to change `portMAX_DELAY` to a reasonable timeout. Then rx_items will be null, the `else` will be taken below, and the correct error code applied.

There is a similar call in `_reset()` for which the timeout is `100 / portTICK_PERIOD_MS` so I tried that. It seems to work well, and the rest of the error handling functions as expected.

DavidAntliff, 2021-01-12:
Thank you for investigating further, and for the PR. Looks good to me.

DavidAntliff, 2021-01-12:
https://github.com/DavidAntliff/esp32-owb/pull/21/files

commit 0d2f9133 (2021-01-12): Report timeout errors when reading/resetting the bus (fixes #19)
commit b2032f71 (2021-01-12): Merge pull request #21 from lukecyca/fix-timeout

Report timeout errors when reading/resetting the bus (fixes #19)
closed 2021-01-12 by DavidAntliff