DavidAntliff/esp32-owb #23: Example

Koxx3, 2021-04-10:
Hello,
this is very interesting !
Do you have an example using RMT please ?
Regards,
F.

DavidAntliff, 2021-04-10:
How about this one? https://github.com/DavidAntliff/esp32-ds18b20

It's an example of the DS18B20 one-wire temperature sensor using this component and RMT. Let me know if you have any questions.

Koxx3, 2021-04-10:
thanks a lot. i'll try tomorrow 😉

Koxx3, 2021-04-11:
hummmm ... unfortunatly, I use Arduino (big project) and it doesn't seems so eay to integrate espressig native SDK parts inside my project :(

DavidAntliff, 2021-04-11:
Unfortunately I can't really provide any advice or support for the use of this within an Arduino project. Maybe someone else can help you - I'm sure other people have managed to wrap native SDK components for use with Arduino. I'll close this issue but feel free to comment further if you need to.

closed 2021-04-11 by DavidAntliff