DavidAntliff/esp32-owb #24: Example schematic

luiz-brandao, 2021-05-05:
Hi there, how should I wire my circuit to use the RMT? It seems like the library uses different GPIOs for TX and RX so I'm a bit confused. Could you provide an example?

Before finding this library I was trying to write my own version but I was having some funny issues using the same pin for TX and RX (switching it back and forward).

Thank you!

DavidAntliff, 2021-05-05:
Hi, I don't have a schematic handy but you might find this useful: https://github.com/DavidAntliff/esp32-ds18b20-example#hardware

Basically you just need to connect power + ground to the device, and then connect a single wire with the appropriate pull-up resistor to the pin you specify in the menuconfig. You only need one pin, and RMT doesn't care which one provided it's a GPIO (there are some that are input-only: 34 to 39 IIRC).

Let me know if you need any more info. If you can find a similar Arduino project for a one-wire device then you can just use the same wiring.

luiz-brandao, 2021-05-05:
Ok, so you can use the _same_ pin for RX and TX. I thought they were in different pins. How do you make it work, you just have to assign different channels for RX and TX?

DavidAntliff, 2021-05-05:
Yes, you can use the same pin for RX and TX. In the library, it switches it from an RMT input to an RMT output automatically. Yes, there are different channels within RMT for each direction (I think, it's been a while since I last looked closely at the code).

luiz-brandao, 2021-05-05:
Thanks. I'm just starting with ESP32 and your code is a great reference!

DavidAntliff, 2021-05-05:
No problem - it's a great little platform, and it's really well supported by Espressif. All the best, and please do let me know if you run into problems with this or any of my other ESP-IDF-based libraries.

I'll close this for now, but feel free to reply now or in future if necessary.

closed 2021-05-05 by DavidAntliff