DavidAntliff/esp32-owb #25: Cache disabled but cached memory region accessed error when writing to flash

maxsydney, 2021-09-21:
I am intermittently encountering this error when writing data to flash. I have done a bit of poking around in the config to see which settings will influence whether this error occurs, but so far I am unable to stop this from happening unless I remove the `ESP_INTR_FLAG_IRAM` flag from `rmt_driver_install`. 

Steps required to produce error

- Active communication over owb rmt (DS18B20 in this case)
- Mounts SPIFFS file system (using `esp_vfs_spiffs_register`)
- Write data to filesystem

This error will occur perhaps 1/10 writes. Presumably this is related to whether an RMT interrupt is firing when the flash is accessed. I have not been able to produce this error when reading from flash, but reads are happening less frequently in my application. 

Any ideas on what might be causing this?

DavidAntliff, 2021-09-21:
Hi, I'm sorry I don't know enough about how the interrupts and FLASH system works on the ESP32 to give you much advice.

I have a couple of other projects that use the RMT, such as the [Frequency Counter](https://github.com/DavidAntliff/esp32-freqcount), so perhaps you could try using that with your FLASH writer and see if you have the same problem?

If so, then I'd suggest contacting Espressif directly, either through their GitHub [Issues](https://github.com/espressif/esp-idf/issues) or via their [forum](https://www.esp32.com/).

DavidAntliff, 2022-02-13:
Hi Max,

I don't know what the implications might be for removing that flag - best to ask the Espressif team directly. I've had good experiences with contacting them through their forum, they do tend to reply in most cases.

If you do ask on the forum, please post or send me a link to the discussion and I'll keep an eye on it also.

maxsydney, 2022-02-15:
Thanks David. You can find the question posted on the forum here https://esp32.com/viewtopic.php?f=2&t=26144

alemariusnexus, 2022-05-26:
Just wanted to say that I'm experiencing the same error. For me, it only seems to happen when compiling with size optimizations (-Os), not with -O2.

I've posted a reply with more info and some stab-in-the-dark findings in the forum thread. It's currently waiting for moderator approval.

DavidAntliff, 2022-07-14:
> Any news?

Nothing that I'm aware of - best bet is probably to try and get one of the Espressif devs into the linked discussion at esp32.com above, see what they think.

If there's any indication that my code is doing something incorrectly and causing this fault then I'm more than happy to fix it, I just don't know what that might be, yet.

kriegste, 2022-07-14:
I'm going to force-inline those functions mentioned (called in the isr) and see and report what happens.

kriegste, 2022-08-05:
The issue is about to be fixed
https://github.com/espressif/esp-idf/issues/9487