DavidAntliff/esp32-owb #26: not compiled with esp32c3

ASRSim13, 2021-12-08:
> ./components/esp32-owb/owb_rmt.c
../components/esp32-owb/owb_rmt.c: In function '_init':
../components/esp32-owb/owb_rmt.c:432:26: error: incompatible types when assigning to type 'volatile union <anonymous>' from type 'int'
         GPIO.enable_w1ts = (0x1 << gpio_num);
                          ^
../components/esp32-owb/owb_rmt.c:436:14: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'enable1_w1ts'; did you mean 'enable_w1ts'?
         GPIO.enable1_w1ts.data = (0x1 << (gpio_num - 32));

DavidAntliff, 2021-12-10:
Which compiler version (`gcc --version`) and ESP-IDF version are you using?

ASRSim13, 2021-12-13:
> Which compiler version (`gcc --version`) and ESP-IDF version are you using?

ESP-IDF v4.3.1
-- Found Git: /usr/bin/git (found version "2.25.1")
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
with other versions - the same problem

DavidAntliff, 2021-12-19:
Thanks for the report, I will look into this when time permits. On the face of it, It's probably a portability issue.

bbinet, 2022-01-28:
I'm also facing the same issue, does not compile on esp-c3:
https://github.com/DavidAntliff/esp32-ds18b20-example/issues/24

MRusinov, 2022-02-17:
Same problem, ESP32C3, ESP-IDF 4.4

DavidAntliff, 2023-03-26:
Thanks to contributor @mjcross there's now support for ESP-IDF v4.4.4 and v5.0.1. If you have time, would you mind trying again with the latest code from the `master` branch, please?

apexskier, 2023-12-17:
https://github.com/CJCombrink/esp32-owb resolved this for me, esp IDF v5.1.2